proveskit/pysquared #9: Implement new IMU Code

Mikefly123, 2024-06-27:
There is a new IMU on the Internal FC Board V1a. We need to add the libraries and initializations.

Mikefly123, 2024-06-28:
In order to make this work, you will need to pull in the following libraries from the Adafruit side: 
https://github.com/adafruit/Adafruit_LSM6DS
https://github.com/adafruit/Adafruit_LSM303_Accel

Mikefly123, 2024-06-28:
# Electrical Hookup for the IMU Components
[image]

Sankie005, 2024-07-02:
Tried giving it a shot, I don't have access to a board or a test platform right now so the results are going to be mixed, however I hope that some parts of the code can be reused, I've Dm'd you the code and explanation instead of creating a new branch

PR #15 (2024-07-13): IMU code
Added code that reads gyro, acceleration, and temperature data. `c.accel` pulls data from the accelerometer while `c.imu_accel` pulls acceleration data from the IMU. Resolves #9.
closed 2024-07-13 by NicoleMaggard