proveskit/pysquared #40: CircuitPython RV3028 Implementation

Mikefly123, 2024-10-14:
# Planned Features
We're adding an `rv3028.py` to the CircuitPython lib folder and folding in use of the Real Time Clock into `pysquared.py`. 

The primary things we want it to accomplish are as follows: 
- [ ] Get and Set the Time
- [ ] Set and Check an Alarm
- [ ] Check for Events Through the EVI Function 
- [ ] Manage the Trickle Charger

# Required Hardware
A FC Board V4b or higher or a BB V3b or higher is required to use the Real Time Clock.

Mikefly123, 2024-10-14:
# Getting and Setting Time
So far, I have already added this functionality in Commit #[be88cd7 ](https://github.com/proveskit/circuitpy_flight_software/commit/be88cd735b0d35ddfd5f73c5761a708cd419120a) it just now needs to be tested! 

You should be able to do the following: 
```py 
from pysquared import cubesat as c

print(c.time) # prints current time
c.time(hours, minutes, seconds) # sets current time

print(c.date) # returns current date
c.date(year, month, day, weekday) # sets current date
```

dbgen1, 2025-01-18:
I holistically tested getting and setting the time. With my changes, the implementation is correct and the driver is fully functional. There was a small issue in the way the property was being utilized, as per the [https://docs.python.org/3/howto/descriptor.html](python docs), the setter property can only take one argument, so this was corrected in #2d028e3. The time and date are accessed the same way, but can now be set using the setter function with:
```python
c.time = (hours, minutes, seconds)
c.date = (year, month, day, weekday)
```

I will use the branch linked to this issue to continue the implementation of the RV3028 in pysquared.

Mikefly123, 2025-01-18:
Thanks for this @dbgen1! You may want to go ahead and do a pull request to get that setter change merged before dev goes too far ahead. Let me know if you need some more implementation notes for how to bring in all the other features. 

Now that I think about it, with the way that Nate has restructured library imports, I think we should make a new repo in the PROVES Kit Organization specifically for a CircuitPython RV3028 Library. We could maybe even contribute it upstream once it is done!