proveskit/pysquared #46: SD card umount function doesn't work on reset_vbus, and reset_vbus kills our satellite

Michael-Esser777, 2024-10-29:
```
 def reset_vbus(self):
        # unmount SD card to avoid errors
        if self.hardware["SDcard"]:
            try:
                umount("/sd")
                self.spi.deinit()
                time.sleep(3)
            except Exception as e:
                self.error_print(
                    "error unmounting SD card" + "".join(traceback.format_exception(e))
                )
        try:
            self._resetReg.drive_mode = digitalio.DriveMode.PUSH_PULL
            self._resetReg.value = 1
        except Exception as e:
            self.error_print(
                "vbus reset error: " + "".join(traceback.format_exception(e))
            )
```
We get an error on the "umount("/sd")" line:
[pysquared]error unmounting SD cardTraceback (most recent call last):
  File "lib/pysquared.py", line 630, in reset_vbus
OSError: [Errno 22] Invalid argument

We've tried everything including manually adding a sd directory to the rp2040. 
Also, even when we comment out the umount line, this function deinitializes our radio and kills the satellite. 
When we comment out the function entirely the satellite works perfectly fine.

Mikefly123, 2024-10-29:
Hey @Michael-Esser777 is this an error that you are getting when `pysquared.py` is initialized or when you try to call this function? 

Also, there aren't any line numbers here, so could you let me know what is on line 630 or upload a copy of your file to the issue ticket?

Michael-Esser777, 2024-10-30:
Line 630 is the umount("/sd") line

We get this error when c.check_reboot() is called in the main loop. 

```
def check_reboot(self):
        self.UPTIME=self.uptime
        self.debug_print(str("Current up time: "+str(self.UPTIME)))
        if self.UPTIME>86400:
            self.reset_vbus
```

We actually discovered this error because we messed with our BOOTTIME causing self.UPTIME to be greater then 86400. My understanding of this code is that the satellite attempts to reboot itself every 86400 seconds, aka every day. However, instead it is deinitializing the radio and freezing the satellite

Mikefly123, 2024-10-30:
Got it! I suspect this might be a case of changed syntax between different versions of CircuitPython. The original PyCubed Flight Software that we based our code on was written way back during CircuitPython 6.0! 

I'll do a little more investigating, for now I would recommend just commenting out or writing `pass` on your local `c.check_reboot()` so you can still test things in the loop.

Mikefly123, 2024-10-31:
@Michael-Esser777 when you get the chance can you send me a screenshot of your terminal when `pysquared` inits? Main thing I would like to see is if the SD card initializes correctly at the beginning.

Michael-Esser777, 2024-11-03:
[image]
[image]

The first image is where the SD card initializes, the second image is the error that is getting thrown. The code freezes immediately after

Mikefly123, 2024-11-09:
Hey @Michael-Esser777 I took another look at this and I noticed that the error message thrown is actually on Line 631 in the screenshot you sent. Is it possible that the error is due to `self.spi.deinit()` being incorrectly called? One thing to note is that the in the newer versions of the flight software we don't have a generic `self.spi` bus and instead have `self.spi1` and `self.spi0` respectively.

Michael-Esser777, 2024-11-11:
The error was on line 630 but then I added a line above before sending that screenshot so now it's on 631. Should we try calling 
self.spi0.deinit() and self.spi1.deinit() at the same time then? Currently its only calling self.spi0.deinit().

Mikefly123, 2024-12-28:
Hi @Michael-Esser777 sorry I we got out of step with helping you on this issue! Your last comment was just a few days before we handed off the Orpheus satellite (which is now in space!) so we lost track of this thread. 

Is this an issue you are still grappling with?

Michael-Esser777, 2024-12-29:
No worries, not a super pressing issue. Yes, this is still an issue were struggling with. Just to recap, I believe the issue is with the code attempting to reboot the satellite after it has been running for a day, however it doesn't work and kills the satellite instead, so right now our solution is just to get rid of that functionality. I can't do any work on this until after break (January 20th), but lmk if you have know of any fixes and I can try them out once I get back.

Mikefly123, 2025-01-04:
So right now, the code that is on the `main` branch of the repo has a working soft reset timer (that just calls `microcontroller.reset()` at the allotted time` that is verified as working on Pleiades - Orpheus in space! 

If you are fine with just that functionality then I would recommend just merging that into your current code base and running with it. It might be a little messy with using that function without unmounting the sd card, but we haven't seen any issues with data corruption on our side as long as you aren't resetting midway through a write operation or something.

Mikefly123, 2025-07-30:
Closing this issue as solved with the current codebase. We may have to take a closer look at this behavior though in future implementations!