proveskit/pysquared #48: Flight Controller Uptime Overflow

Mikefly123, 2024-11-01:
Didn't have a clock running but we see an uptime overflow on the flight controller board. This is a problem, the board was on for maybe only 15 minutes. 
[image]

blakejameson, 2024-12-18:
Yesterday, I tried reproducing the issue on the Orpheus branch and was unable to reproduce the overflow. Was this issue addressed?

Mikefly123, 2024-12-28:
We have also been unable to replicate this issue. It is possible that it was a one off variance with that board. Maybe an issue with the internal clock? 

I am going to close this issue for now. Maybe someday we will encounter it again.

closed 2024-12-28 by Mikefly123