proveskit/pysquared #56: Move all Orpheus code/hard coded strings into config file options

nateinaction, 2024-12-19:
Read a config file at boot that can:
- name the satellite
- set the radio callsign
- set all the jokes that can be beaconed
-

Alandlt15, 2024-12-22:
On the branch 'config-file', I created the config file and added the stuff mentioned. 
Not sure if that's exactly how its supposed to go so feel free to correct me if needed.

nateinaction, 2024-12-24:
This is a good start. I like that you're isolating these config lines into their own file. I saw your comments `# might need to config these as well` I think that's a good idea for this PR. Go ahead and tackle at least the rest of that block of variable declarations!

For the next phase in moving these hard coded values to config what about using something like [configparser](https://docs.python.org/3/library/configparser.html) or [json](https://docs.python.org/3/library/json.html#module-json)? It looks like both are built-in python libraries included in python 3.0 so they _should_ be supported in circuit python. I'm leaning toward json since it is more common than INI. What do you think?

Alandlt15, 2024-12-25:
I will go ahead and get the rest of the variable declarations migrated. 

And absolutely! I'm on board with using json although I'm gonna do some research and try to find out the pros and cons of both if any.

closed 2025-01-05 by Alandlt15