proveskit/pysquared #71: Occasional restart: Satellite has no attribute 'can_bus'

nateinaction, 2025-01-04:
The board is occasionally restarting after receiving the following errors:
[image]
[image]

Doesn't happen on every reboot. Smells like a race condition.

Mikefly123, 2025-01-06:
Hmm very interesting! The `uptime` being 2953s indicates that this is not an issue related to cold starting the board. 

I did notice that you dropped a `TODO` on the jury rigged can bus sleep call in in `pysquared.py`:
```py
            # TODO(nateinaction): Accessing private method. Ability to sleep the canbus should be made in upstream.
            self.can_bus._set_mode(0x20)
```
Based on the traceback my guess is that if the CAN transceiver is already in sleep mode and then we try to command it to go to sleep mode again it can trigger the timeout `RuntimeError` when it tries to set sleep mode the second time. My guess is that usually this is not a problem, unless this error occurs when trying to setup the can_bus object for the second time after a soft refresh of the board (i.e. using `CTRL+D` in the REPL)! This creates a compounding unhandled `AttributeError` in main when `functions.safe_sleep()` tries to sleep the `can_bus` but no such object exists. 

@NicoleMaggard this may be what is causing the double booting error that we saw in the Orpheus telemetry! Good catch @nateinaction, if this bug explanation sounds good to you, I think we can close this issue as an understood bug and mark as `wontfix` due to our plans to remove the CAN bus from V2 PROVES Kit.

nateinaction, 2025-01-07:
I've had a couple questions about this. Let's leave the issue open so we can point to it until the offending code is removed.

nateinaction, 2025-01-07:
@Mikefly123 Hmmm I'm working on one of the new v4c boards you sent to @blakejameson and this is occurring on every boot. Do you think we can go ahead and remove the can bus code?

Mikefly123, 2025-01-07:
@nateinaction I think its fair to go for it and depreciate CAN bus support at this time. Especially if it greatly smooths out the developer experience. 

As a sidebar though, I think I need to setup somewhere on the docs page to announce errata with our boards. On FC Board V4cs with serial numbers #0018-#0028 the CAN Bus Transceiver does not function at all. Which I now realize will cause this error to throw on every boot.

Mikefly123, 2025-01-07:
Note: I believe errata on the #0018-#0028 boards is unrelated to the original error this issue was opened for, because that other error also seems to occur on earlier serial numbered FC boards.

nateinaction, 2025-01-07:
Resolved by #80

closed 2025-01-07 by nateinaction