proveskit/pysquared #75: Remove boot rollover code

nateinaction, 2025-01-06:
pysquared tracks how many boot cycles a system has gone through by storing a count in a byte in persistent memory. We believe that the count will automatically rollover after it hits 255 but we have added a safety catch that forces a rollover at 200: 
https://github.com/proveskit/circuitpy_flight_software/blob/84674e0afd75da4caa3dc66055925c825e28a93f/FC_Board/lib/pysquared.py#L207-L208

To improve code flow and reduce complexity we could remove this rollover code if someone proves that we don't need it...

**Thoughts on how to test:**
Create a simple/blank main.py, on boot set a byte in persistent memory to 255, add 1 to that byte and set it, read the byte. If the byte reads 0 and no errors are found, we should be able to remove this code.

Mikefly123, 2025-01-06:
I concur on this. For whomever decides to tackle this I believe you can also run the test from the `REPL` using the following lines of code: 
```py
from pysquared import cubesat as c #If not already imported

c.c_boot = 255
print(c.c_boot)

c.c_boot += 1
print(c.c_boot)
```

Alandlt15, 2025-01-07:
I gave this a shot using Michael's suggestion by using the REPL. 

After adding 1 to c.c_boot, it kept staying at 255. I'm a little unsure on what I should do next.

nateinaction, 2025-01-07:
It looks like `bitflags` is a library that we created. It is possible to add automatic rollover to the entire library if that's the behavior we prefer. Are we using this anywhere else that would expect different behavior?

https://github.com/proveskit/circuitpy_flight_software/compare/reorg-fc_board-to-root...bitflagrollover?expand=1#diff-e3516aa414af49bb3696ce619921f6e94007c64d7e26589d3e3e6ffe43ccacfdL47-R47

nateinaction, 2025-01-18:
Completed in #87. Closing.

closed 2025-01-18 by nateinaction