proveskit/pysquared #77: Remove the Camera Code from pysquared.py

Mikefly123, 2025-01-06:
# What Needs to be Done
Right now, we have the `ov5640` library loaded into `pysquared.py` and setup as a `self.cam` object. This was a sort of stop gap way for us to try and shove a camera into the Pleiades - Orpheus mission and PROVES Kit V1.5. This implementation is rife with problems though, so we will be scrapping it in V2 PROVES Kit and opting instead for a Sony Spresense based camera solution in work by the Northeastern and Columbia teams. 

At this time, it would be great if someone could go through and scrub all references to the `ov5640`, its `cam` object, and any related data structures it uses from the code base. 

## Example of the Code
```py
        """
        Camera Initialization
        """
        if self.hardware["TCA"] is True:
            try:
                self.cam: adafruit_ov5640.OV5640 = adafruit_ov5640.OV5640(
                    self.tca[5],
                    data_pins=(
                        board.D2,
                        board.D3,
                        board.D4,
                        board.D5,
                        board.D6,
                        board.D7,
                        board.D8,
                        board.D9,
                    ),
                    clock=board.PC,
                    vsync=board.VS,
                    href=board.HS,
                    mclk=None,
                    shutdown=None,
                    reset=None,
                    size=adafruit_ov5640.OV5640_SIZE_QVGA,
                )

                self.cam.colorspace = adafruit_ov5640.OV5640_COLOR_JPEG
                self.cam.flip_y = False
                self.cam.flip_x = False
                self.cam.test_pattern = False

                self.cam.effect = 0
                self.cam.exposure_value = -2
                self.cam.white_balance = 2
                self.cam.night_mode = False
                self.cam.quality = 20

                self.hardware["CAM"] = True

            except Exception as e:
                self.error_print(
                    "[ERROR][CAMERA]" + "".join(traceback.format_exception(e))
                )
                self.hardware["CAM"] = False
```

Alandlt15, 2025-01-07:
Okay started working on this on the branch that's tagged on this issue. I started by just commenting out the lines in relation to the things you mentioned, tested on the board and it seems to run just fine. 

A question I have is, would I be able to also remove the adafruit_ov5640 library as its only used in the code that needs to be removed?

Mikefly123, 2025-01-07:
Hey @Alandlt15 yeah, I think the `adafruit_ov5640` library is the only one that needs to be removed. Really straightforward overall I think.

closed 2025-01-08 by blakejameson