proveskit/pysquared #79: Remove can_bus From the Codebase

Mikefly123, 2025-01-07:
# Why is This Change Being Made?
A CAN bus transceiver was introduced in the V1.5 PROVES Kit as a means of facilitating board to board communication between the RP2040 mounted on the battery board and the RP2040 mounted on the flight controller board. This never really panned out, with the preferred method of doing board to board communication being UART for lower power consumption and simplicity. As a result in PROVES Kit V2 we will be phasing out use of the `can_bus`. 

Due to recurring problems with the `can_bus` (as seen in #71) it may be more convenient at this point to toss the `can_bus` out of the code base rather than backtracking to fix it just to remove it later. 

# What Needs to Be Done? 
The `can_bus` needs to be purged from the code base. This includes mentions of it in `pysquared.py`, `functions.py` and libraries and helper files such as `adafruit_mcp2515` and `can_bus_heplper.py`. Attached as some (but not all) examples of where this code lives in the codebase: 

In `pysquared.py`:
```py
        """
        CAN Transceiver Initialization
        """
        try:
            self.spi0cs2: digitalio.DigitalInOut = digitalio.DigitalInOut(
                board.SPI0_CS2
            )
            self.spi0cs2.switch_to_output()
            self.can_bus: CAN = CAN(self.spi0, self.spi0cs2, loopback=True, silent=True)
            self.hardware["CAN"] = True
            # TODO(nateinaction): Accessing private method. Ability to sleep the canbus should be made in upstream. Search for this comment to find other usages.
            self.can_bus._set_mode(0x20)

        except Exception as e:
            self.debug_print(
                "[ERROR][CAN TRANSCEIVER]" + "".join(traceback.format_exception(e))
            )
```
In `functions.py`: 
```py
    def safe_sleep(self, duration: int = 15) -> None:
        self.debug_print("Setting Safe Sleep Mode")

        # TODO(nateinaction): Accessing private method. Ability to sleep the canbus should be made in upstream. Search for this comment to find other usages.
        self.cubesat.can_bus._set_mode(0x20)
```

blakejameson, 2025-01-07:
@Mikefly123 
[image]

It looks though can_bus_helper is currently only being used in the Batt_Board, which will soon be removed from the repo. Since Batt_Board will be removed, there will be no calls to it. Instead of removing the can_bus from its existence in can_bus_helper.py, should I just delete can_bus_helper.py instead?

Mikefly123, 2025-01-07:
Hey @blakejameson, yeah I think you can just go ahead and do that!

blakejameson, 2025-01-07:
Thank you @Mikefly123 !

[image]

Similarly, adafruit_mcp2515 is included mostly in Batt_Board and when it is included in pysquared.py as 'CAN', I removed where CAN is used. 

Instead of modifying the adafruit_mcp2515 regarding the can_bus, would I be able to go ahead and remove adafruit_mcp2515, as it will no longer be needed?

Mikefly123, 2025-01-07:
He @blakejameson, yeah you can indeed go ahead and just remove the whole `adafruit_mcp2515` library. That library is the driver for the CAN bus transceiver chip that's on the board.