proveskit/pysquared #91: Does changing RP2040 clock speed change file read/write times?

nateinaction, 2025-01-10:
## Summary
It takes about 170 seconds to load the flight software onto the board with `make install`. Can it be done faster? We reduce the CPU clock speed on the satellite as a power saving feature, let's see what kind of impact it has on development speed.

## Suggested testing methodology

1. Delete all files from the board
2. Use screen to drop into the repl on the board:
  ```sh
  screen /dev/tty.usbmodem101
  ```
3. Set the clock speed high
  ```python
  machine.set_clock(125000000)  # 125Mhz
  ```
4. From your computer, test read/write speed
  ```sh
  # cd to the board mount
  cd /Volumes/PYSQUARED
  # Test board write speed
  dd if=/dev/zero of=./test bs=5k count=10 oflag=direct
  ```
5. Set clock speed low
  ```python
  machine.set_clock(62500000)  # 62.5Mhz
  ```
6. From your computer, test read/write speed
  ```sh
  # cd to the board mount
  cd /Volumes/PYSQUARED
  # Test board write speed
  dd if=./test of=/dev/zero bs=5k count=10 oflag=direct
  ```
7. Post the results in the issue

Board mount location may differ. You can find out how to find the volume in the `README.md`.

Mikefly123, 2025-01-13:
@yudataguy I think based on my read of that console output there is a very minor reduction in read/write speeds when we drop the clock speed? Only a difference of 349 bytes/s which isn't too much, but maybe you'd still feel it as a couple seconds over the course of a few megabytes?

Mikefly123, 2025-01-15:
I think that power consumption is not really a big concern when we are just copying files over on the ground. I figure a good amendment to the code would be to default the clock speed to the standard 125Mhz when it is on the ground and only change it down to the 62.5Mhz when we are going for flight and looking to save power. 

Perhaps an amendment to `pysquared.py` along the lines of this:
```py
if seld.debug:
    self.turbo=True
else:
    self.turbo=False
```

Mikefly123, 2025-02-08:
Hey @nateinaction do you think we should just update `pysquared.py` to just use the default clock speed of `125Mhz`? It seems like there is a marginal improvement in the file copy times if we do this. 

Trying to close some of the stale issues haha

nateinaction, 2025-02-11:
If it's only a marginal improvement, let's keep it in "flight mode." Going to close. Feel free to reopen.

closed 2025-02-11 by nateinaction