proveskit/pysquared #102: Add Radio Config to config.json

Mikefly123, 2025-01-20:
# Proposed Change
We currently have a dictionary in `pysquared.py` that stores values for configuring the radio settings. We should try and extract these to be contained in `config.json` for better modularity and easier parameter management! 
```py
        self.radio_cfg: dict[str, float] = {
            "id": 0xFB,
            "gs": 0xFA,
            "freq": 437.4,
            "sf": 8,
            "bw": 125,
            "cr": 8,
            "pwr": 23,
            "st": 80000,
        }
```

Alandlt15, 2025-01-28:
@Mikefly123, do you happen to know what the full 'name' for these keys are? I'm trying to put as much 'human-readable' values in the json file and I'm a little bit confused on what gs, sf, bw, cr, pwr, and st mean.  

Additionally, for the commands in cdh.py, Nate and I talked about how that dictionary might be better protected inside of the class, rather than in the json file. Simply because we imagine that these commands will and should not be modifiable at run-time. Could this be the same for 'id' and 'gs'? Or any other one? 

Thank you.

Mikefly123, 2025-01-28:
@Alandlt15 oh yeah sorry about that! 
- `id`: The 1 byte ID value for the radio you are transmitting from. Also known as the `node` in the Radiohead Format
- `gs`: The 1 byte ID value for the radio you are transmitting to (the Ground Station). Also known as `destination` in the Radiohead Format
- `freq`: The radio frequency in Mhz (Megahertz)
- `sf`: The radio `spreading_factor`
- `bw`: The radio `bandwidth` in Khz (Kilohertz)
- `cr`: The radio `coding_rate`
- `pwr`: The `transmit power` for the radio.
- `st`: Genuinely have no idea what this one is, it is a legacy thing that we should probably investigation. I think it is a `Start Time` or something?

Mikefly123, 2025-01-28:
Re: The command keys in `cdh.py`

Yes I concur that those should probably be left hard coded rather than modifiable during runtime. I think we actually want to leave all of the other radio parameters open to change during runtime if possible, so in the future we can tweak the inflight radio performance as needed.

Alandlt15, 2025-01-28:
Got it got it. I appreciate the clarification!

closed 2025-01-29 by Alandlt15