proveskit/pysquared #103: Aligning Virginia Tech InspireFly Code with Main

Mikefly123, 2025-01-21:
# Virginia Tech InspireFly
The InspireFly team at Virginia Tech are scheduled to launch and deploy their 1U selfiesat from the ISS sometime in 2025. 

They have a codebase that was forked from this repo pre-V1.0.0 and David Encarnacion has asked us to take a look and see if we can help them get up to speed with the latest! 

https://github.com/daveencar03/circuitpy_flight_software_WITH_MODIFICATIONS_-DAVE-3-

nateinaction, 2025-01-22:
Hmmm challenging... Can we enumerate what features they added and where they added them to? After we have that list maybe we can choose which features we want to adopt and decide how to merge them?

daveencar03, 2025-01-22:
Yeah, sure thing!

The only new feature I have added directly to my knowledge is the [async def pcb_comms()](https://github.com/daveencar03/circuitpy_flight_software_WITH_MODIFICATIONS_-DAVE-3-/blob/3365176067f0d198b48e3db8e62d4a18ac7e754b/main.py#L263C1-L300C37), along with it's call as t7 in [main_loop()](https://github.com/daveencar03/circuitpy_flight_software_WITH_MODIFICATIONS_-DAVE-3-/blob/3365176067f0d198b48e3db8e62d4a18ac7e754b/main.py#L302C1-L315C29). There are also some external imports like [easy_comms_circuit ](https://github.com/daveencar03/circuitpy_flight_software_WITH_MODIFICATIONS_-DAVE-3-/blob/3365176067f0d198b48e3db8e62d4a18ac7e754b/main.py#L16) and [FCB_class](https://github.com/daveencar03/circuitpy_flight_software_WITH_MODIFICATIONS_-DAVE-3-/blob/3365176067f0d198b48e3db8e62d4a18ac7e754b/main.py#L18).

Mikefly123, 2025-01-22:
Hm, interesting! A few notes as we think about how to tackle this: 
- In V1.0.0 we tossed out using `asyncio` to create a bunch of tasks in `main.py` when we upgraded to using the new `adafruit_rfm` library rather than the `pysquared_rfm9x` there were weird race conditions being created by having `asyncio` in use both at the library level and the main code level. The code now just runs a linear path through all of the possible tasks. 
- The changes made look pretty straightforward actually and relatively generic. @daveencar03 if you're able to join one of our upcoming software meetings to explain how it is supposed to work it might make sense for us to adopt your board to board comms library into the main software. 
- This does raise an interesting question for us though about how we want to handle development of "core" features of the kit and how to support mission specific features. Something like a board to board comms driver sounds pretty core, but what if in the future we have an oddly specific implementation of something, how would be handle that? 

The next software working group meeting on Thursday at 4:30PM would be a great time to discuss this!

daveencar03, 2025-01-22:
I may be a tad late, I get out of a class at ~4:45, however, I will do my best to attend if it's all the same!

daveencar03, 2025-01-22:
Oh perfect! Then yes, I'll be in attendance! Appreciate the clarification

Mikefly123, 2025-03-06:
Closing this issue as the Virginia Tech team will be proceeding with their custom implementation of the flight software: 

https://github.com/daveencar03/circuitpy_flight_software_WITH_MODIFICATIONS_-DAVE-3-

closed 2025-03-06 by Mikefly123