proveskit/pysquared #110: Add GitHub Issue and PR templates

nateinaction, 2025-01-25:
## Story
**As a** PROVES Kit code reviewer
**I want** PR and Issue templates that describe requirements for those interactions
**So that** I more easily review those documents

## Acceptance Criteria
- Issues templates
  - Bug report template
    - Summary
    - Reproduction steps
  - New feature template
    - Story
      - **As a**
      - **I want**
      - **So that**
    - Acceptance Criteria
    - Technical Details
- PR template
  - Summary
  - How was this tested
    - [ ] Added new unit tests
    - [ ] Ran code on the hardware (screenshots are helpful)
    - [ ] Other (Please describe)

## Technical Details
- GitHub offers a way to [encourage users to create useful issues and PRs](https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests) via Pull request and Issue templates. 
- Here's a collection of issue templates created by others for some ideas. Are there any other things in here that you think would be helpful to have in these templates? https://github.com/stevemao/github-issue-templates

JamesDCowley, 2025-01-30:
Created a [pull request] (https://github.com/proveskit/circuitpy_flight_software/pull/123) for this. Changing the name to be lowercase may work.

JamesDCowley, 2025-01-30:
Last PR didn't seem to work, going to try and move the PR template file out of its own directory and put it in /.github

JamesDCowley, 2025-01-30:
That last PR worked 👍

[image]

closed 2025-01-30 by JamesDCowley