proveskit/pysquared #128: Minor Updates Through Overwriting `config.json`

Mikefly123, 2025-01-31:


Alandlt15, 2025-03-08:
@Mikefly123 I plan to start working on this over the break and hopefully I can get some good progress done. As this past Monday, at the lab we played around with the USBFunctions found in lib.pysquared.usb. Ultimately, we were able to create files in the sd card and even edit these files through the use of the usb functions. 

Later that day, we to also edited the current config.json file through the board from the repl which was really cool and amazing to see. Since we now know what steps we need to take for the board to write to its own file system, I'm starting to look at implementing the command to overwrite config.json and I have some questions about it.

1. What is the process of creating/adding a command and how do we hone in on a specific byte array for the command?
 
2. I'm not sure if this is a good question, but what is the encoding/decoding preference for our satellite? 

3. This one might be like question 2, but when I print this out for one of the commands `print(b"\xd4\x9f".decode("utf-8", errors="ignore"))` I get 'K' as output, and when I print this `print(int.from_bytes(b"\xd4\x9f", "big"))` I get 54431, I'm not sure if this matters or not?

Mikefly123, 2025-03-10:
Hey @Alandlt15 a quick answer to your questions:

1. To add a command, you simply add a function to the `cdh.py` that you would like to be executed when the bytecode is called in the commands dictionary. There is no system right now for defining the byte arrays, so you can use whatever you want. We probably should standardize command codes at some point in the future though.
2. There is currently no preference, but David and Madhav are working on a system right now for standardizing operational codes in our packet headers. You may want to catch up with their work!
3. No that doesn't matter, because the `cdh.py` is just checking if any two bytes are like any other two bytes they are not limited to ASCII characters. In that first example there happens to be an ASCII representation of that byte code, but in the second one you are getting the integer representation of the byte code (don't you love Python's mutability haha)

Alandlt15, 2025-03-10:
Awesome! Thank you for taking the time to answer I'll definitely get in touch with David and Madhav to see how their work is so far. 

Lol heck yea! Python being Python.