proveskit/pysquared #139: [FEATURE] Set RTC Time on `make install`

Mikefly123, 2025-02-04:
## Story
* **As a** satellite developer.
* **I want** for my board to know the correct time.
* **So that** logs and other time sensitive information is correctly timestamped. 

Currently our log files are all being timestamped with a nebulous mystery time that the microcontroller is keeping. Wouldn't it be cool for this to be the actual real world time? 

## Acceptance Criteria

- [ ] Establish a method for `make install` to automatically open a serial connection with the flight controller board and enter the REPL
- [ ] Call the `rtc` time and date setter functions to set the RTC time to the current UTC time. 

## Technical Details
On flight controller board V4b and above there is an RTC (Real Time Clock) IC that is onboard for keeping time! There are still some features not yet introduced into the drivers (like sustaining the time after being powered down and wake up alarms), but within a single power up the time does seem to be kept pretty correctly!

nateinaction, 2025-02-04:
Right now `make install` only needs to know about the filesystem mount point. We would need to make the command also know about the tty/serial port. I suspect the behavior will also be different between the posix systems and windows.

Maybe we could start with a separate `make sync-time` command to feel out an implementation?

Mikefly123, 2025-02-04:
Yeah, I could see an independent command making sense! I'll update the original ticket to reflect that.

blakejameson, 2025-02-11:
I tried to utilize the RTC methods but experienced the bug discussed in https://github.com/proveskit/PROVES_CircuitPython_RV3028/issues/4

[image]

blakejameson, 2025-03-14:
Hey @Mikefly123, Quick question:

In https://github.com/proveskit/circuitpy_flight_software/issues/170#issuecomment-2686467817, you mention "check the rtc current time and if it is more than a few seconds off, it should sync and then notify the user that the time was off by x amount and needed to be synced."

Should the acceptance criteria of this issue be updated to include that? Or should this issue only deal with setting the time when the `make sync-time` command is called and a subsequent issue would deal with detecting the difference of times and only updating if the difference in times is large enough?

blakejameson, 2025-03-18:
@Mikefly123 Sorry to @ you again but you may have missed this previous question I asked. Let me know if you get the chance

Mikefly123, 2025-03-18:
Thanks for the reminder on this @blakejameson! I think for this issue we should just treat `make sync-time` as a function that always resets the RTC time to the current time regardless of whether or not the time is already in sync or is not in sync.

Easier to implement and I think we would do the drift checking mentioned in #170 as a part of the ground station software down the line

blakejameson, 2025-03-28:
As a note, sync-time functionality now lives in https://github.com/proveskit/sync-time. through the help of uv, sync-time is brought into the v4 and v5 boards and used in the `make sync-time` command

blakejameson, 2025-03-29:
Closing this issue, as https://github.com/proveskit/CircuitPython_RP2040_v4/pull/3 was approved by Nate and merged in after he tested the command

closed 2025-03-29 by blakejameson