proveskit/pysquared #141: [TASK] Reimplement Colorful Print Statements

Mikefly123, 2025-02-04:
## Story
* **As a** `circuitpy_flight_software` developer.
* **I want** colorful print statements in the terminal.
* **So that** I can quickly determine the severity and location of the `info`, `debug`, and `error` messages that appear in the terminal. 

One of the features that was recently lost when we moved to `json` format logging is colorful print statements in the terminal. Colorful print statements were useful for intuitive debugging (like highlighting error prints in **red**) and locating where particular debug messages were originating (by having different classes print in different colors). 

If we reimplement colorful inputs, I think at a minimum we would like to see highlighting of logs that are `error` severity be red and maybe `warning` be orange or something.

## Acceptance Criteria

- [ ] Modify `logger.py` to use colorful prints based on logging level.
- [ ] Verify that colorful prints don't interfere with any other functionalities.

## Technical Details
Here is a Copilot generated explanation of using the `debug_color` method in the `pysquared` library: 

To use the `debugcolor` module to change the color of `print` statements, you need to import the `co` function from `debugcolor` and use it to format your message with the desired color. Here is an example:

1. Import the `co` function:
```python
from debugcolor import co
```

2. Use the `co` function to format your message with the desired color and print it:
```python
print(co(msg='hello!', color='red'))
```

The `co` function takes three parameters:
- `msg`: The message you want to print.
- `color`: The color of the text. Available options are "red", "green", "orange", "blue", "pink", "teal", "white", and "gray".
- `fmt`: The format of the text. Available options are "normal", "bold", and "ulined" (underlined). The default is "normal".

Example with different colors and formats:
```python
print(co(msg='This is red text.', color='red'))
print(co(msg='This is green bold text.', color='green', fmt='bold'))
print(co(msg='This is blue underlined text.', color='blue', fmt='ulined'))
```

This will print the messages in the specified colors and formats.

blakejameson, 2025-02-05:
Just a comment, but the team filled me in that Niko from Mt. SAC is going to work on this issue. I don't think they have been added to the proveskit org yet, but after they are, they can officially be assigned on here

Nikko-Adrian-Pacleb, 2025-02-05:
My solution for this one is I made a LogsColor Dictionary that calls the function co
```py
LogColors = {
    "NOTSET": "NOTSET",
    "DEBUG": co(msg="DEBUG", color="blue"),
    "INFO": "INFO",
    "WARNING": co(msg="WARNING", color="orange"),
    "ERROR": co(msg="ERROR", color="pink"),
    "CRITICAL": co(msg="CRITICAL", color="red"),
}
```

Then I replaced the string after the json.dumps function (I added a coloredJsonOuput variable just in case we need the jsonOutput somehwere else)
```py
json_output = json.dumps(kwargs)

        if self._can_print_this_level(level_value):
            colored_json_output = json_output.replace(f"\"level\": \"{level}\"", f"\"level\": \"{LogColors[level]}\"") # or maybe use level_value
            print(colored_json_output)
```

blakejameson, 2025-02-05:
As Nikko doesn't have a board yet, I told him I would run his code on my board
This is the following output from running the code:
[image]

closed 2025-03-04 by Nikko-Adrian-Pacleb