proveskit/pysquared #146: [Proposed TASK] Implement `circup` for CircuitPython Library Updates

Mikefly123, 2025-02-05:
## Story
* **As a** `circuitpy_flight_software` developer.
* **I want** to have my libraries up to date.
* **So that** we don't miss bug fixes or other updates to our underlying software libraries.

I recently discovered there is a CircuitPython package manager from Adafruit for autoupdating libraries called [circup](https://github.com/adafruit/circup). We have a pretty good system for doing it now, but if it would be useful in our build system I wanted to bring it up! 

## Acceptance Criteria

- [ ] Perhaps creating a `make lib-update` command that uses `circup` to update the libraries on the target board? 

## Technical Details
From the GitHub: 

> Each CircuitPython library on the device usually has a version number as metadata within the module.
>
> This utility looks at all the libraries on the device and checks if they are the most recent (compared to the versions found in the most recent version of the Adafruit CircuitPython Bundle and Circuitpython Community Bundle). If the libraries are out of date, the utility helps you update them.

nateinaction, 2025-02-07:
I considered using `circup` when working on #66 but found the purpose of that tool too specific for what I was hoping to provide. The `circup` description explains
> This utility looks at all the libraries on the device and checks if they are the most recent (compared to the versions found in the most recent version of the Adafruit CircuitPython Bundle and Circuitpython Community Bundle). If the libraries are out of date, the utility helps you update them.

While we do store versions of libraries on the board, our main goal is to store those same versions on the developer's machine so the IDE can provide hinting for them. In addition, we want to make sure that we are shipping consistent library versions to every PROVES Kit user. To do that we need the ability to declare specific library versions using configuration like a `requirements.txt` stored in git. At the time `pip` seemed like a natural candidate and, with the added requirement for managing a developer's python environment, now `uv`.

`circup` seems like a neat tool and something I kind of want to nab some functionality from (like automatic board detection for the `make install` command) but I don't think it's the right tool for maintaining library versions for this project.

What do you think?

Mikefly123, 2025-02-07:
Hey Nate! Yeah this all sounds good to me, sticking to `uv` for now makes a lot of sense. I do agree with you that borrowing the automatic board detection stuff would be pretty great. 

I'll close this issue for now. It was good that this thinking was documented here!

closed 2025-02-07 by Mikefly123