proveskit/pysquared #150: [TASK] Refactor Sleep Functions

Mikefly123, 2025-02-07:
## Story
* **As a** `circuitpy_flight_software` developer
* **I want** the ability to safely sleep the satellite when needed
* **So that** we can sleep the satellite without being reset by the watchdog timer

Right now the satellite hardware (with the watchdog timer enabled) will be reset if the `watchdog_pin` is not toggled every 26s. If we want the satellite to go idle and sleep for longer than 26s we need dedicated functions to periodically come out of sleep and pet the watchdog. 

## Acceptance Criteria

- [ ] `safe_sleep()` refactored from `functions.py` into something like a `sleep_helper` class
- [ ] `short_hybernate()` and `long_hybernate()` also refactored
- [ ] [Stretch] Implement support for waking the satellite from a hibernate on `pin_alarm` from the `rtc` (Real Time Clock) 

## Technical Details
You may want to consider reading about RP2040 deep sleep on the [Adafruit Docs](https://learn.adafruit.com/deep-sleep-with-circuitpython/rp2040-sleep). 

### Functions to Refactor
These are in `functions.py` 

```py
    def safe_sleep(self, duration: int = 15) -> None:
        self.logger.info("Setting Safe Sleep Mode")

        iterations: int = 0

        while duration > 15 and iterations < 12:
            time_alarm: circuitpython_typing.Alarm = alarm.time.TimeAlarm(
                monotonic_time=time.monotonic() + 15
            )

            alarm.light_sleep_until_alarms(time_alarm)
            duration -= 15
            iterations += 1

            self.cubesat.watchdog_pet()

...

    def Short_Hybernate(self) -> Literal[True]:
        self.logger.debug("Short Hybernation Coming UP")
        gc.collect()
        # all should be off from cubesat powermode

        self.cubesat.enable_rf.value = False
        self.cubesat.f_softboot.toggle(True)
        self.safe_sleep(120)

        self.cubesat.enable_rf.value = True
        return True

    def Long_Hybernate(self) -> Literal[True]:
        self.logger.debug("LONG Hybernation Coming UP")
        gc.collect()
        # all should be off from cubesat powermode

        self.cubesat.enable_rf.value = False
        self.cubesat.f_softboot.toggle(True)
        self.safe_sleep(600)

        self.cubesat.enable_rf.value = True
        return True
```

blakejameson, 2025-02-07:
@Mikefly123 Would `short_hybernate()` and `long_hybernate()` go into the `sleep_helper` class as well? Also, `listen_loiter` uses `safe_sleep` so I am wondering if it would be moved into the `sleep_helper` class

blakejameson, 2025-02-07:
^ @Mikefly123 Nevermind to the question about `listen_loiter()`. The function also uses the `listen()` function which is present in functions.py. It looks like `listen_loiter()` stays in functions.py

Mikefly123, 2025-02-07:
Yeah, I think the hybernate functions should come to the `sleep_helper` as well!

blakejameson, 2025-02-07:
@Mikefly123 Quick question, but for Short_Hybernate and Long_Hybernate, I was curious if this was a typo, as Hibernate is the usual spelling. But it turns out that Hybernate is an obsolete spelling of hibernate. I didn't know if this was some inside joke or whatever the case may be. Let me know if Hybernate should stick or if I should instead update it to Hibernate

[image]

Mikefly123, 2025-02-07:
You can feel free to update the spelling. I think back in the day Nicole and I just liked the spelling with a `y` better haha

closed 2025-02-11 by blakejameson

blakejameson, 2025-02-11:
One more note:

Regarding the stretch goal for this Issue, once in @nateinaction 's office hours, I asked @dbgen1 for guidance about how to tackle the stretch goal, as it involves the RTC. He thought that the driver would have to be modified to allow for this functionality, and I brought up the idea that the stretch goal could potentially be something he tackled in a later issue

Mikefly123, 2025-02-11:
Hey @blakejameson that makes sense to me, thanks for the update!