proveskit/pysquared #156: [BUG] <Error when self.legacy is set to True>

blakejameson, 2025-02-09:
## Summary
When `legacy` is set to 'true' in the config.json, this can cause an error. The Satellite has a property called `self.legacy`, whose boolean value is dictated by the config.json.

During the Satellite initialization process, a field called self.enable_rf is set to be a digitialIO object if `self.legacy` is True. However, as shown in screenshot below, a `board.RF_ENAB pin` in the firmware is attempted to be used. 
[image]

When the `board.RF_ENAB` pin is not found, this results in an error that looks like the following:
[image]

Micheal filled me in on the error in an earlier PR: https://github.com/proveskit/circuitpy_flight_software/pull/151#issuecomment-2646626040

## Steps to reproduce
change `legacy` to equal 'true' in the config.json and run the code
[image]

## Environment

Mikefly123, 2025-02-09:
Just to add a little bit more context to this, if you are using a `legacy` configuration of the PROVES Kit (i.e. a V4 Flight Controller Board with a V2 Battery Board like the InspireFly Team) the pinout of the Flight Controller board needs to be completely different than what it is if you are using the `latest` configuration of the PROVES Kit (V4 Flight Controller Board with a V3 Battery Board). 

In the `latest` configuration the flight controller board does not have access to the voltage regulator that supplies power to the radios, because of that there is no pin on the board that is configured as `board.RF_ENAB`. In `legacy` configurations and in the upcoming V5 Flight Controller Board we are planning to return this functionality so the flight controller board is able to directly turn on or off the power to the radio module.

Mikefly123, 2025-06-27:
Closing this as `self.legacy` isn't a thing anymore!

closed 2025-06-27 by Mikefly123