proveskit/pysquared #157: Unused functions and variables

blakejameson, 2025-02-11:
## Summary
There are some variables and functions that are defined but never utilized. 

Earlier I stopped by the Discord Meeting Room and asked @Mikefly123  if I could run the unused variables and functions by him to learn which should stay vs. which are no longer needed

## Files and their unused functions, variables
**battery_helper.py**
Functions currently not used anywhere:
•	*Currently there are no BatteryHelper function calls anywhere in the repo

**cdh.py**
Functions currently not used anywhere:
•	hotstart_handler
•	FSK

**Field.py**
Functions currently not used anywhere:
•	troubleshooting

**functions.py**
Variables not being used anywhere:
•	self.error_count (error_count was replaced by an nvm Counter object that is stored in the Logger)
•	self.detumble_enable_x
•	self.detumble_enable_y
•	self.detumble_enable_z

Functions currently not used anywhere:
•	current_check
•	send_packets
•	listen_joke
•	get_battery_data
•	OTA

**packet_sender.py**
Functions currently not used anywhere:
•	handle_retransmit_request
•	fast_send_data

**pysquared.py**
Functions currently not used anywhere:
•	reset_vbus
•	powermode
•	print_file
•	read_file
•	new_file

Mikefly123, 2025-02-11:
Okay, here's a start at outlining what some of these unused functions do!

**battery_helper.py**
This was a helper class for interacting with the battery board. Note how there are a bunch of functions in there just supporting board to board UART communications. We could just nuke the whole thing, or leave it in and just leave it for now and refactor it for the upcoming V2 updates.

**cdh.py**
- `hotstart_handler` this is a function that indeed is currently unused, but is intended as a helper function to ease the satellite back into normal operations after a hibernation period. I would leave it for now until the `rtc`and `radio` library updates are done and we reimplement genuine hibernation functionality.
- `FSK` I believe this is in to assist in switching between `lora` and `fsk` modes on the satellite. I think this should be refactored to align with Nate's new radio modulation manager.

Mikefly123, 2025-02-12:
Some more explanation of the unused functions: 

**Field.py**
- troubleshooting: I think this function can just be removed, I think its an artifact of when we first setup that class. Eventually we will refactor this class into some kind of packet factory.

**functions.py**
- `error_count`: since the way we track this has been changed, I think we can remove this function from `functions.py`
- All the `detumble_enables`: These variables are in here so that we could easily decide to toggle the enable state of the x, y, or z, magnetorquers for detumbling. Since we are going to have to do a general refactor of detumble capabilities anyways, feel free to remove these
- `current_check()`: This was intended as a helper function for managing the power mode of the satellite from the `functions.py`. As seen in #159 I want to refactor this to its own interface, so it can be removed from here.
- `send_packets()`: This function accepts data larger than 250 Bytes and breaks it up into individual packets that are send by the `packet_sender` class. It should stay in as an option to be used, but maybe will be refactored into the future packet factory.
- `listen_joke()`: This is a legacy function from the Yearling satellites that allows for you to tell the satellite a joke (after which it will laugh at you). This can probably be refactored into `cdh.py` at some future time.
- `get_battery_data()`: This function will have to be completely refactored for PROVES Kit V2. It should probably actually go into the interface described in #159. 
- `OTA()`: This is a placeholder function for doing over the air updates. A key feature in the V2.0.0 software we're working towards

blakejameson, 2025-02-14:
@Mikefly123 Thank you so much for taking the time to sift through all of these functions and variables! I really appreciate it Michael

blakejameson, 2025-02-26:
Just reopened issue, as there are still functions that haven't yet been removed

blakejameson, 2025-03-07:
@Mikefly123 Looking to potentially close this issue but first wanted to clear with you if `handle_retransmit_request` and `fast_send_data` should continue to stay around or should be removed

Mikefly123, 2025-03-08:
Hey Blake, I think we should just leave that stuff as is until we do a complete rework of sending packetized data through the radio.

blakejameson, 2025-03-08:
@Mikefly123 Thank you Michael, sounds good. Going to go ahead and close this issue

closed 2025-03-08 by blakejameson