proveskit/pysquared #158: [TASK] Update `lib` Requirements.txt to Use Our Own TCA Driver

Mikefly123, 2025-02-12:
## Description of the Task
This should be really straightforward. All we need to do is update the `requirements.txt` in the `lib` folder of the repo to use our modified version of the `Adafruit_CircuitPython_TCA9548A` instead of the the upstream version. The reason we have a modified library is because of a very unsafe `while` loop in the upstream library as described in #104. 

## Acceptance Criteria
- Verify that the new modified library loads when `make install` is run instead of the original library.
- [Optional but Preferred] Implement Unit Tests for the Functions that Call the TCA
- Check with @dbgen1 if there are still updates needed to be made to stabilize our modified library.

## Technical Details
Our version of the driver can be found here:

- https://github.com/proveskit/Adafruit_CircuitPython_TCA9548A

blakejameson, 2025-02-14:
@Mikefly123 I went ahead and added the proveskit repo's TCA9548A to the requirements

However, regarding the unit tests optional criteria, I think it may be the case where it's difficult to unit test the TCA. Currently, I see that the TCA is passed as an argument to some classes, and the TCA is then used in other functions in those classes. I am not sure if things are extracted in a way to test the TCA specifically
Let me know your thoughts

Mikefly123, 2025-02-15:
Hey @blakejameson! Yeah, I am also not too sure about how we would implement testing for the TCA. Let's think about it for a day or two, and if there isn't anything sensible we can come up with we should just push ahead for now I think. It might be something that can only really be tested in the loop.

The reason the TCA object gets passed around quite a bit is that once it is instantiated it acts like an `i2c` object for the devices that need to call sensors connected through the TCA.

blakejameson, 2025-02-17:
@Mikefly123 Have spent a bit of time trying to think if anything would come to me regarding testing the TCA, but haven't had any success. Are you still on board with pushing ahead with this?

Mikefly123, 2025-02-17:
@blakejameson Yeah let's just get this merged for now to fix the bug and we can think about testing later

closed 2025-02-17 by blakejameson