proveskit/pysquared #159: [TASK] Create a `state_of_health` Interface to Control Power Modes

Mikefly123, 2025-02-12:
## Story
* **As a** `circuitry_flight_software` developer.
* **I want** a simpler way to interact with the satellite's state_of_health.
* **So that** we have a single source of truth of the current health of the satellite systems to determine which mode we should be in.

The `state of health` interface of the satellite should serve two roles, collecting metadata (like temperatures of key components and the voltage of the battery) and changing the current operational mode of the satellite as a result of what this meta data tells us. Implementing a `mode_manager` should probably be a separate task. 

## Data Points to Collect
All of these data points should be collected by the module and saved to a single data structure.
- [ ] System Voltage 
- [ ] System Current
- [ ] Solar Voltage
- [ ] Solar Current
- [ ] Radio Temperature
- [ ] Microcontroller Temperature
- [ ] Internal Temperature
- [ ] Error Count

> NOTE: If you are currently testing on a V4 flight controller board without a battery board there is no way to collect the voltage and current information. This might be a bit of a blocker if there is no easy way to just ignore the fact that those sensors are not connected.

## Acceptance Criteria

- [ ] Clean Out All of the Old State of Health Functions and Attributes
- [ ] Consolidate state of health collection into a single interface module
- [ ] Implement unit tests on the state of health system (example: checking that mode switching is done correctly)

## Technical Details
A great place to start with this is in `pysquared.py`, where we have this function that sets the `power_mode` and manipulates some of the hardware as a result of the current `mode`: 

```py 
    def powermode(self, mode: str) -> None:
        """
        Configure the hardware for minimum or normal power consumption
        Add custom modes for mission-specific control
        """
        try:
            if "crit" in mode:
                self.neopixel.brightness = 0
                self.enable_rf.value = False
                self.power_mode: str = "critical"

            elif "min" in mode:
                self.neopixel.brightness = 0
                self.enable_rf.value = False

                self.power_mode: str = "minimum"

            elif "norm" in mode:
                self.enable_rf.value = True
                self.power_mode: str = "normal"
                # don't forget to reconfigure radios, gps, etc...

            elif "max" in mode:
                self.enable_rf.value = True
                self.power_mode: str = "maximum"
        except Exception as e:
            self.logger.error(
                "There was an Error in changing operations of powermode",
                err=e,
                mode=mode,
            )
```
The `power_mode` attribute is then checked in `main.py` to determine what functional flow the satellite should take at any given time:
```py
    ######################### MAIN LOOP ##############################
    try:
        while True:
            # L0 automatic tasks no matter the battery level
            c.check_reboot()

            if c.power_mode == "critical":
                c.RGB = (0, 0, 0)
                critical_power_operations()

            elif c.power_mode == "minimum":
                c.RGB = (255, 0, 0)
                minimum_power_operations()

            elif c.power_mode == "normal":
                c.RGB = (255, 255, 0)
                main()

            elif c.power_mode == "maximum":
                c.RGB = (0, 255, 0)
                main()

            else:
                f.listen()
```

JamesDCowley, 2025-02-21:
Hey @Mikefly123, can you help me with getting data for the `solar_volage` and `solar_current`? Not sure how to proceed on getting these values. Also, can you look at how I'm getting the current values. `system_voltage` and `system_current` return `null`. Not 100% sure I'm getting the right values. This is currently how I'm getting each one.

```python
            self.state["system_voltage"] = self.c.micro.cpu.voltage  # not 100% sure on this
            self.state["system_current"] = self.c.charge_current
            self.state["solar_voltage"] = None # unsure on how to get these values
            self.state["solar_current"] = None   
            self.state["battery_temperature"] = self.f.last_battery_temp # literally just gets a value from config  
            self.state["battery_voltage"] = self.c.battery_voltage
            self.state["radio_temperature"] = self.f.last_radio_temp()
            self.state["microcontroller_temperature"] = self.c.micro.cpu.temperature
            self.state["internal_temperature"] = self.c.internal_temperature
            self.state["error_count"] = self.logger.get_error_count()
            self.state["power_mode"] = self.c.power_mode
            self.state["uptime"] = self.c.get_system_uptime
            self.state["boot_count"] = self.c.boot_count.get()
            self.state["burned_flag"] = self.c.f_burned.get()
            self.state["brownout_flag"] = self.c.f_brownout.get()
            self.state["fsk_flag"] = self.c.f_fsk.get()
```
And this is what it returns:
```{"msg": "State of Health", "time": "2020-01-01 00:27:54", "state": {"solar_current": null, "microcontroller_temperature": 23.8615, "internal_temperature": 22.0664, "error_count": 181, "brownout_flag": true, "battery_voltage": null, "system_current": null, "radio_temperature": 24, "power_mode": "normal", "uptime": 1674, "fsk_flag": false, "solar_voltage": null, "battery_temperature": 20.0, "system_voltage": null, "burned_flag": true, "boot_count": 115}, "level": "INFO"}```

Thanks!

Mikefly123, 2025-02-21:
Hey @JamesDCowley these values are usually called from the battery board. In the upcoming V5 Flight Controller you'll have direct access to these values but for now there isn't a good way of getting that data without hooking into a Battery Board. 

Here's an example of what it looked like in the `pysquared_eps` that is now in the [software archive(https://github.com/proveskit/software-archive).

```py
    # =======================================================#
    # Getters for State of Health Monitoring                #
    # =======================================================#
    @property
    def battery_voltage(self):
        if self.hardware["PWR"]:
            voltage = 0
            try:
                for _ in range(50):
                    voltage += self.pwr.bus_voltage
                return voltage / 50 + 0.2  # volts and corection factor
            except Exception as e:
                self.debug_print(
                    "[WARNING][PWR Monitor]" + "".join(traceback.format_exception(e))
                )
        else:
            self.debug_print("[WARNING] Power monitor not initialized")

    @property
    def system_voltage(self):
        if self.hardware["PWR"]:
            voltage = 0
            try:
                for _ in range(50):
                    voltage += self.pwr.bus_voltage + self.pwr.shunt_voltage
                return voltage / 50  # volts
            except Exception as e:
                self.debug_print(
                    "[WARNING][PWR Monitor]" + "".join(traceback.format_exception(e))
                )
        else:
            self.debug_print("[WARNING] Power monitor not initialized")

    @property
    def current_draw(self):
        if self.hardware["PWR"]:
            idraw = 0
            try:
                for _ in range(50):  # average 50 readings
                    idraw += self.pwr.current
                return idraw / 50
            except Exception as e:
                self.debug_print(
                    "[WARNING][PWR Monitor]" + "".join(traceback.format_exception(e))
                )
        else:
            self.debug_print("[WARNING] Power monitor not initialized")

    @property
    def is_charging(self):
        return not (self.charge_indicate.value)

    @property
    def charge_voltage(self):
        if self.hardware["SOLAR"]:
            voltage = 0
            try:
                for _ in range(50):
                    voltage += self.solar.bus_voltage
                return voltage / 50 + 0.2  # volts and corection factor
            except Exception as e:
                self.debug_print(
                    "[WARNING][SOLAR PWR Monitor]"
                    + "".join(traceback.format_exception(e))
                )
        else:
            self.debug_print("[WARNING] SOLAR Power monitor not initialized")

    @property
    def charge_current(self):
        if self.hardware["SOLAR"]:
            ichrg = 0
            try:
                for _ in range(50):  # average 50 readings
                    ichrg += self.solar.current
                return ichrg / 50
            except Exception as e:
                self.debug_print(
                    "[WARNING][SOLAR PWR Monitor]"
                    + "".join(traceback.format_exception(e))
                )
        else:
            self.debug_print("[WARNING] SOLAR Power monitor not initialized")
```

JamesDCowley, 2025-02-21:
@Mikefly123, should I implement these in `pysquared` so I can access to them in `state_of_health`? Is there an equivalent to `hardware["PWR"]` and `hardware["SOLAR"]` in the current repo? I notice that these values exist but are all set to `None` here:

[image]