proveskit/pysquared #164: [BUG] <RTC Error when Year Exceeds 159>

blakejameson, 2025-02-18:
## Summary
Using the real time clock, setting the year to a number larger than 159 causes an Exception for `rtc.set_date`

As shown in code below, setting the year to 159 was fine, but 160 caused an Exception
[image]

Here are the 2 pertinent lines of code run in main.py to test this. The first line runs fine but the second causes the Exception
[image]

## Steps to reproduce
Put the code shown above in main.py and run the board normally

## Environment

blakejameson, 2025-02-19:
@dbgen1 
Alright, so I just tried to explore the error some

With the '_int_to_bcd', passing the argument of 159 results in a value of 249.

With the '_int_to_bcd', passing the argument of 160 results in a value of 256.

Bytes object can only take a range up until 255, so it makes sense that the 256 will cause an error

blakejameson, 2025-02-19:
@dbgen1 All that being said, it seems like the code can stay as is, and someone could just pass 25 as the value to signify '2025'

blakejameson, 2025-02-19:
Omg, what a huge fail by me. All I had to do that whole time was scroll down to get_date where it says a value between 0 and 99 😢 😭 
[image]

I am sorry for all the time wasted here

dbgen1, 2025-02-20:
I'll do a PR to update this so there is bounds checking and error messages when data that is out of range is passed. That should make the behavior a little more understandable. Thanks for the report!

closed 2025-02-28 by dbgen1