proveskit/pysquared #170: [FEATURE] Utilize RTC to generate timestamp in Logs

blakejameson, 2025-02-19:
## Story
* **As a** `circuitpy_flight_software` user
* **I want** to be able to see an accurate current time and date in log output
* **So that** I am able to know at the precise time events occur

Currently, the logger outputs a time and date that is based on time.localtime(). 
Having the logger use the RTC instead would allow for accurate time and date output.

## Acceptance Criteria

- [ ] Uses the RTC to retrieve the current time and date
- [ ] Logs output with a timestamp based on RTC 

## Technical Details

The Logger class will be refactored to utilize the RV3028 module

blakejameson, 2025-02-20:
@Mikefly123 In the process of working on this issue, I realized that the RTC object is created in the Satellite object in pysquared.py

Of course the Logger would need to access the RTC object, but being that the Satellite has a parameter it takes in of Logger, the Satellite object can't then be passed into the Logger object

A potential solution is initializing the RTC and I2C1, as the RTC depends on it, in main.py. Then passing the RTC to the Logger, and the I2C1 and RTC to Satellite

I was curious on what your thoughts were with this approach or if you recommend another way to go about it

Mikefly123, 2025-02-20:
Hey @blakejameson! I think a good way around this issue is to use the built in RTC that the RP2040 has as the means of time stamping the logs: 

https://docs.circuitpython.org/en/latest/shared-bindings/rtc/

This RTC would not have an way of sustaining itself between resets (our external RTC theoretically would, but the library support for that is not done and tested yet), so you would need to resync it after every reset and probably also periodically check if the time is drifting (internal microcontroller clocks are notorious for drifting over time).

blakejameson, 2025-02-21:
@Mikefly123 Thanks again for sending me this link! Using the built in RTC has been a huge help already

Currently in the early stages but can now enter the REPL to set the time and date and then control-D to soft reset and I see the proper time and date values for log output timestamps.

As you can see at the bottom of the output, the check_reboot() function results in an uptime of a huge number (162,247,351 seconds in this case) that far exceeds the 3600 second reboot time set in the config. This of course causes a hard reset

It looks like I will get to refactoring the BOOTTIME and other related logic to prevent this. 

It makes sense that the uptime currently shows as a huge number, as the current time now is far beyond the initial value of 1577836800 (January 1, 2020) 

[image]

blakejameson, 2025-02-26:
Update:

I finally solved an error in which I had spent around 8 hours over a few days. 
The error was that after updating the RTC to the current time in the REPL, the first time a `safe_sleep` was occurred and thus the `alarm.light_sleep_until_alarms(time_alarm)` was called, it would hang indefinitely.

I somehow managed to get past this error tonight and now make it past the alam.light_sleep calls

Anyways, in comparing the RP2040's RTC vs my computer's Clock app, it appears it is doing well (though this hasn't been running too long)

For this test so far, I set the RTC to the then-current time of 7:29pm. Later at 8:05:52pm, the RP2040 RTC was very close, as shown at the bottom of serial output

[image]

blakejameson, 2025-02-26:
> Hey [@blakejameson](https://github.com/blakejameson)! I think a good way around this issue is to use the built in RTC that the RP2040 has as the means of time stamping the logs: [...]

@Mikefly123 Regarding periodically checking if the time is drifting, I am struggling to think of a way to do this programmatically. Do you have any suggestions or guidance on how to do this? Or would checking if time drift is present be a manual task and someone would simply just run the function to update the time again?

Mikefly123, 2025-02-26:
Hey @blakejameson! Great work on sifting through this issue!

Just to summarize what we just discussed on call:
- We should have `make sync-time` check the `rtc` current time and if it is more than a few seconds off, it should sync and then notify the user that the time was off by `x` amount and needed to be synced. This should be sufficient for now
- Once we dive into the mission operations software side of things we'll want to come up with some kind of function that will allow us to remotely sync the time over the radio. Probably a quick packet exchange between the satellite and ground station that compares the time between the two things and compensates for time of flight and stuff

blakejameson, 2025-03-02:
@Mikefly123 Regarding this issue, I just tried to test the RP2040's RTC again and this time it got fast.

Set the RTC time to the current time of 4:14 CST and then 24 minutes later, it was around 2 seconds too fast. 48 minutes after the start, it was 5 seconds too fast. Then 56 minutes after the initial manual sync, it was around 7 or 8 seconds too fast. I will show those results below.

To clarify, as `make sync-time` is a separate issue and the one I will work on next, the work you just mentioned above will be in that issue as opposed to this one, correct? Do I have the green light to put in a PR for this issue right here?

[image]

[image]

[image]

Mikefly123, 2025-03-04:
Interesting! That seems like quite a high level of drift over the course of about just one hour. Do you happen to remember what the temperature of the room was while you were running this test? I know clocks can have a major dependency on the ambient temperature for maintaining time. I made this attached plot real quick to show that it seems like the drift is exponential.

[image]

Go ahead and make the pull request for now, but I would recommend noting this drift (maybe in a `readme` about the logger functions?) and opening a new ticket for something to do further testing on how much the RTC will become unsynced over time. 

Thanks!

blakejameson, 2025-03-06:
@Mikefly123 Sorry for the late reply but I do not know what the temperature was in the room during that trial. I am going to go open a new ticket related to the RTC drift tests you suggested

blakejameson, 2025-03-06:
An hour test. Fared better than last time. Stayed on time until the 50 minute mark, when it was 1 second faster than the world time
[image]

closed 2025-03-07 by blakejameson