proveskit/pysquared #174: [BUG] Missing Satellite class attributes

SamuelGomezG, 2025-02-21:
## Summary
There are two functions in `cdh.py` that are trying to access non-existent attributes of the `Satellite` class in `pysquared.py`. What should I do in these cases?
https://github.com/proveskit/circuitpy_flight_software/blob/9ad507456fab462c900e1c71583510b4e2270aff/lib/pysquared/cdh.py#L112
https://github.com/proveskit/circuitpy_flight_software/blob/9ad507456fab462c900e1c71583510b4e2270aff/lib/pysquared/cdh.py#L176

Mikefly123, 2025-02-21:
Hey @SamuelGomezG! Good catch on these two. For the Line 176 `f_hotstrt` flag we can remove that. We don't currently have any hot start logic in the flight software. 

For the Line 112 `c_gs_resp` this looks like it is trying to access a counter for how many messages have come in from the ground station. This could be useful, but we don't currently have a use for it. You can probably just remove for now and we can reimplement it if it ends up being needed.

SamuelGomezG, 2025-02-21:
Hello, @Mikefly123. Thanks for replying!

I removed the problematic lines in my refactor branch. Will do a PR soon.

closed 2025-02-21 by SamuelGomezG