proveskit/pysquared #186: Testing the RP2040 RTC's time drift

blakejameson, 2025-03-06:
Following Michael's advice in https://github.com/proveskit/circuitpy_flight_software/issues/170#issuecomment-2696294119, I am going to do trials to test the performance of the RP2040's RTC and see how much the time becomes out of sync with time

blakejameson, 2025-03-07:
An hour test from earlier at 2:15pm. Fared better than last time. Stayed on time until the 50 minute mark, when it was 1 second faster than the world time

[image]

blakejameson, 2025-03-13:
Surprisingly stayed on time for the whole hour
[image]

blakejameson, 2025-03-14:
Another great time trial
[image]

Mikefly123, 2025-03-26:
I think these tests indicate that there shouldn't be any problem with RTC drift at this time. Thanks for collecting this data Blake, it is great documentation!

I added a link to this issue on the docs site in case we want to come back to it someday.

closed 2025-03-26 by Mikefly123