proveskit/pysquared #198: [CLEANUP] repl test files

blakejameson, 2025-03-17:
## Summary
Currently, there are several repl tests in the repo. We were curious to know which of these are important going forward and if there are any tests that are no longer relevant or worth having in the repo

As a circuitpy_flight_software developer, 
I want to know which REPL tests are relevant, 
so I have a clear understanding

blakejameson, 2025-03-18:
@Mikefly123 We are curious if you have any thoughts and insight regarding the repl test files currently present

Mikefly123, 2025-03-18:
Hey @blakejameson! 

Of the tests currently in that directory honestly on the `radio_test` and `LiDARtest` are things are want to carry over into the refactored REPL interactions. The rest of the tests in there are fragments of scripts our team has used over the years to try and validate certain features of the satellite that need to be rethought to be more useful.

Mikefly123, 2025-06-27:
Closing this as the recent #240 updates have replace the `radio_test` functionalities. We should probably still reintroduce `LiDARtest` at some point.

closed 2025-06-27 by Mikefly123