proveskit/pysquared #203: Min-Max Ranges for Config Values

Alandlt15, 2025-03-21:
Hello team! I'm currently working on issue #128 and adding a new command so a mission operator can change values temporary or permanently at runtime. 

But we need to really make sure the new values are secure and do not harm the satellite, so I'm just looking to find the min and max range for the values that have -1 and 1; excluding bool types. (Michael provided me with some ranges but it was very improptu, so some might be incorrect):

[image]

[image]

Mikefly123, 2025-03-26:
Hey @Alandlt15! Here's a typed up list of the values you're still looking for:

- super_secret_code (cdh command password): type[bytes], min[1], max[24]
- repeat_code: type[bytes], min[1], max[4]
- last_battery_temp: Delete this, it should be in Jame's State of Health interface rather than config
- normal_charge_current: min[0.0], max[2000.0]
- normal_battery_voltage: min[6.0], max[8.4]
- critical_battery_voltage: min[5.4], max[7.2]
- battery_voltage: Move to State of Health Interface
- current_draw: Move to State of Health Interface
- sleep_duration: min[1], max[86400]
- normal_battery_temp: min[1], max[35]
- normal micro_temp: min[1], max[50]
- sender_id: min[0], max[255]
- start_time: min[0], max[80000]
- transmit_frequency: min0[435], max0[438], min1[915], max1[915]
- ack_delay: min[0.0], max[2.0]
- braodcast_address: min[0], max[255]
- node_address: min[0], max[255]
- modulation_type: min[0], max[1]

Alandlt15, 2025-03-26:
Thank you so much!