proveskit/pysquared #205: [BUG] Stuck petting watchdog

blakejameson, 2025-03-24:
## Summary
I noticed that after 3 minutes and 42 seconds into running, the main loop is blocked by the watchdog pet and the "petting watchdog" output is the only thing that is output 

The following output occurred for 6 minutes more

[image]

## Steps to reproduce

## Environment

Mikefly123, 2025-03-25:
Hey @blakejameson, can you verify the config of your board when you ran into this issue? iirc you were testing on a V4b, I was unable to reproduce this bug on a V4c using `CircuitPython 9.1.4`.

See the logs from my test run of the `main` branch here:

```json
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
{"time": "2020-01-01 00:02:10", "level": "INFO", "msg": "Booting", "published_date": "November 19, 2024", "software_version": ""}
{"time": "2020-01-01 00:02:10", "level": "INFO", "msg": "Code Starting in 5 seconds"}
{"time": "2020-01-01 00:02:11", "level": "INFO", "msg": "Code Starting in 4 seconds"}
{"time": "2020-01-01 00:02:12", "level": "INFO", "msg": "Code Starting in 3 seconds"}
{"time": "2020-01-01 00:02:13", "level": "INFO", "msg": "Code Starting in 2 seconds"}
{"time": "2020-01-01 00:02:14", "level": "INFO", "msg": "Code Starting in 1 seconds"}
{"time": "2020-01-01 00:02:15", "level": "DEBUG", "msg": "Initializing Config"}
{"time": "2020-01-01 00:02:15", "level": "DEBUG", "msg": "Booting up!", "boot_time": "1577836935s"}
{"time": "2020-01-01 00:02:15", "level": "DEBUG", "msg": "Initializing hardware component", "hardware_key": "I2C0"}
{"time": "2020-01-01 00:02:15", "level": "DEBUG", "msg": "Hardware component not initialized", "cubesat": "Orpheus", "hardware_key": "I2C0"}
{"time": "2020-01-01 00:02:15", "level": "DEBUG", "msg": "Initializing hardware component", "hardware_key": "SPI0"}
{"time": "2020-01-01 00:02:15", "level": "DEBUG", "msg": "Initializing hardware component", "hardware_key": "I2C1"}
{"time": "2020-01-01 00:02:15", "level": "DEBUG", "msg": "Initializing hardware component", "hardware_key": "UART"}
{"time": "2020-01-01 00:02:15", "level": "DEBUG", "msg": "Initializing hardware component", "hardware_key": "IMU"}
{"time": "2020-01-01 00:02:15", "level": "DEBUG", "msg": "Initializing hardware component", "hardware_key": "Mag"}
{"time": "2020-01-01 00:02:15", "level": "DEBUG", "msg": "Initializing hardware component", "hardware_key": "RTC"}
{"time": "2020-01-01 00:02:15", "level": "DEBUG", "msg": "Initializing hardware component", "hardware_key": "SD Card"}
{"time": "2020-01-01 00:02:17", "level": "ERROR", "msg": "There was an error initializing this hardware component", "err": ["Traceback (most recent call last):\r\n  File \"lib/pysquared/pysquared.py\", line 77, in wrapper\r\n  File \"lib/pysquared/pysquared.py\", line 134, in init_sd_card\r\nOSError: no SD card\n"], "hardware_key": "SD Card"}
{"time": "2020-01-01 00:02:17", "level": "DEBUG", "msg": "Initializing hardware component", "hardware_key": "NEOPIX"}
{"time": "2020-01-01 00:02:17", "level": "DEBUG", "msg": "Initializing hardware component", "hardware_key": "TCA"}
{"time": "2020-01-01 00:02:17", "level": "ERROR", "msg": "There was an error initializing this hardware component", "err": ["Traceback (most recent call last):\r\n  File \"lib/pysquared/pysquared.py\", line 77, in wrapper\r\n  File \"lib/pysquared/pysquared.py\", line 155, in init_tca_multiplexer\r\n  File \"lib/adafruit_tca9548a.py\", line 122, in __init__\r\nValueError: No TCA9548A detected at 0x77.\n"], "hardware_key": "TCA"}
{"time": "2020-01-01 00:02:17", "level": "WARNING", "msg": "TCA not initialized"}
{"time": "2020-01-01 00:02:17", "level": "DEBUG", "msg": "PySquared Hardware Initialization Complete!"}
{"time": "2020-01-01 00:02:17", "level": "WARNING", "msg": "Unable to initialize hardware device", "device": "I2C0", "status": false}
{"time": "2020-01-01 00:02:17", "level": "INFO", "msg": "Successfully initialized hardware device", "device": "SPI0", "status": true}
{"time": "2020-01-01 00:02:17", "level": "INFO", "msg": "Successfully initialized hardware device", "device": "I2C1", "status": true}
{"time": "2020-01-01 00:02:17", "level": "INFO", "msg": "Successfully initialized hardware device", "device": "UART", "status": true}
{"time": "2020-01-01 00:02:17", "level": "INFO", "msg": "Successfully initialized hardware device", "device": "IMU", "status": true}
{"time": "2020-01-01 00:02:17", "level": "INFO", "msg": "Successfully initialized hardware device", "device": "Mag", "status": true}
{"time": "2020-01-01 00:02:17", "level": "WARNING", "msg": "Unable to initialize hardware device", "device": "SDcard", "status": false}
{"time": "2020-01-01 00:02:17", "level": "INFO", "msg": "Successfully initialized hardware device", "device": "NEOPIX", "status": true}
{"time": "2020-01-01 00:02:17", "level": "WARNING", "msg": "Unable to initialize hardware device", "device": "WDT", "status": false}
{"time": "2020-01-01 00:02:17", "level": "WARNING", "msg": "Unable to initialize hardware device", "device": "TCA", "status": false}
{"time": "2020-01-01 00:02:17", "level": "WARNING", "msg": "Unable to initialize hardware device", "device": "Face0", "status": false}
{"time": "2020-01-01 00:02:17", "level": "WARNING", "msg": "Unable to initialize hardware device", "device": "Face1", "status": false}
{"time": "2020-01-01 00:02:17", "level": "WARNING", "msg": "Unable to initialize hardware device", "device": "Face2", "status": false}
{"time": "2020-01-01 00:02:17", "level": "WARNING", "msg": "Unable to initialize hardware device", "device": "Face3", "status": false}
{"time": "2020-01-01 00:02:17", "level": "WARNING", "msg": "Unable to initialize hardware device", "device": "Face4", "status": false}
[... cut ...]
{"time": "2020-01-01 00:05:39", "level": "DEBUG", "msg": "Petting watchdog"}
{"time": "2020-01-01 00:05:40", "level": "DEBUG", "msg": "Petting watchdog"}
{"time": "2020-01-01 00:05:41", "level": "DEBUG", "msg": "Petting watchdog"}
{"time": "2020-01-01 00:05:42", "level": "DEBUG", "msg": "Petting watchdog"}
{"time": "2020-01-01 00:05:43", "level": "DEBUG", "msg": "Petting watchdog"}
{"time": "2020-01-01 00:05:44", "level": "DEBUG", "msg": "Petting watchdog"}
{"time": "2020-01-01 00:05:45", "level": "DEBUG", "msg": "Petting watchdog"}
```

blakejameson, 2025-03-26:
Closing due to the revert commit PR being merged in
https://github.com/proveskit/pysquared/pull/209

closed 2025-03-26 by blakejameson