proveskit/pysquared #216: [FEATURE] Add Command to `cdh.py` to Reset the Satellite with a New Initial Code File

Mikefly123, 2025-03-26:
## Story
* **As a** mission operator
* **I want** to change what is the first code file executed on startup
* **So that** over the air updates can be commanded.

## Acceptance Criteria

- [ ] Adds a command to `cdh.py`
- [ ] New command function in `cdh.py` that wraps a `supervisor.set_next_code_file(str agrs)` call
- [ ] Demo of command being accepted by software and changing to a temp file that blinks the Neopixel LED 5 times

## Technical Details
When we upload new code to the satellite, one of the safer ways to try running that update is to use the built in CircuitPython `supervisor` module. This module allows you to define what code file will execute after a reset (instead of the default `code.py` or `main.py`. 

In this feature request, we want to just demo some barebones functionality of using this module, where we add a command to `cdh.py` that will accept a `string` augment that is then passed to a `supervisor.set_next_code_file()` call.

See the API docs at this link:
https://docs.circuitpython.org/en/latest/shared-bindings/supervisor/index.html#supervisor.set_next_code_file

blakejameson, 2025-03-29:
@Mikefly123 Just curious how adamant you are about having the "file that blinks the Neopixel LED 5 times" be part of the acceptance criteria vs showing off another functionality with the use of `supervisor.set_next_code_file`

Am currently running into the following error when trying to create the NeoPixel object in the temp file. 
[image]

I am trying to think of ways to get around the error

Mikefly123, 2025-03-31:
Hey @blakejameson I'm not super adamant, we can change it to be some other demonstration if it makes more sense.

In the screenshot you shared how is that initial file being called? Is the code just being reloaded or does it go through a full reset? If somehow pysquared is sticking around when the new neopixel file is called I can see how you'd get an "in use" error.

blakejameson, 2025-03-31:
@Mikefly123 Using your question, I was able to realize what I was doing wrong and just got the 5 blinks to work

Previously, I called a function from the repl to try to set_next_code_file and I pressed control-D. That resulted in the error. However, running supervisor.reload() after the function call allowed it to work! Thank you for leading me in the right direction to figure it out!