proveskit/pysquared #234: [FEATURE] <title>

blakejameson, 2025-04-14:
## Story
* **As a** developer
* **I want** to type check my code
* **So that** I can ensure the code is more reliable

## Acceptance Criteria

- [ ] Introduce a type checker that is checking several files
- [ ] Type checker runs automatically in CI

## Technical Details
Currently, there is no type checking be done. Type checking could allow us to ensure we are using variables and functions in a correct manner

nateinaction, 2025-04-15:
I have started a branch with some new tooling using pyright: https://github.com/proveskit/pysquared/tree/try-out-pyright

When on that branch you can run `make typecheck` to try it out. There are ~350 identified errors so could be a good weekend project.

Mikefly123, 2025-05-07:
Hey @blakejameson I think this was achieved with #239? If so this is another successful ticket we can close!

Mikefly123, 2025-07-14:
Closing this as completed in #239