proveskit/pysquared #236: [FEATURE] Improve sleep_helper functions

blakejameson, 2025-04-17:
## Story
* **As a** satellite user
* **I want** the ability to sleep longer periods of time (3+ minutes)
* **So that** I can conserve power when needed

## Summary

This issue is built with the help of Michael in https://github.com/proveskit/pysquared/pull/151#issuecomment-2813786198

In the current `sleep_helper` functions, the safe sleep duration is maxed out at 3 minutes. The refactor would allow for sleeping longer than 3 minutes. Also, there is currently logic employed that results in 15-second increments. Example: If specified to sleep for 16 seconds, it will sleep for 30 seconds. The refactor would allow in this instance the satellite to sleep for the specified duration of 16 seconds.

## Acceptance Criteria

- [ ] Satellite able to sleep longer than 3 minutes
- [ ] Set a config field `longest_allowable_sleep_time' that dictates the max sleep duration allowed
- [ ] Satellite able to sleep for specified duration rather than 15-second increments
- [ ]  If possible, use Davit's RTC time alarm functionality

blakejameson, 2025-04-18:
A test of running 2 consecutive 10-minute safe sleeps on a v4b board
[image]
[image]

blakejameson, 2025-04-18:
safe_sleep(20) now sleeps for 20 seconds, rather than previously where it slept for 30 seconds
[image]

blakejameson, 2025-04-18:
@Mikefly123 

I am a bit confused when it comes to the idea of using the alarm now from the RTC instead of the alarm.time.TimeAlarm. I notice that the RV 3028 alarm takes minute, hour, and weekday variables. Being that we'd want the alarm to be specified for a certain number of seconds later, I don't know if the external RTC can be used. Please let me know if I am thinking of things incorrectly 
[image]

Mikefly123, 2025-04-19:
Hey @blakejameson we would ideally use the RV 3028 for sleeping for very long periods of time, like many hours or even days (in extreme situations). This is because we can trust the RV 3028 to be more accurate over a long period of time and it consumers nano amps rather than milliamps when keeping track of time.

CircuitPython has a built in functionality that allows us to [deep sleep until a pin alarm](https://docs.circuitpython.org/en/latest/shared-bindings/alarm/#alarm.exit_and_deep_sleep_until_alarms) which I think could be used in conjunction with the RV3028 alarm functionality to wake the satellite at a particular time and date. This functionality has not been tested, but theoretically when an alarm is set and triggers it should cause a pin change on the (board.rtc_int) pin on the PySquared boards.

I think it would be productive to create two options for sleeping, one for short periods of time (which is in seconds and still pets the watchdog) and one for deep sleep (which is in hours or until a particular time and date). If you think it would be productive to split this functionality into an entirely new helper class (like `hibernation_helper` or something) I think there is a case for that.

blakejameson, 2025-04-30:
@Mikefly123 

I notice that currently the `sleep_duration` is set to 30 seconds, but the output says 'Sleeping for 20 seconds'

[image]
[image]

Would you prefer 20 or 30 seconds for the default `sleep_duration`? Or should I get rid of sleep_duration and just hardcode a number like the following:

[image]

This function above currently uses safe_sleep(config.sleep_duration)

Just curious on the direction I should head

Mikefly123, 2025-04-30:
Hey @blakejameson I think 30s is a good default. I would recommend, if possible, passing the sleep duration into the `debug` statement like this so it's more dynamic as well:
```py
self.logger.debug(f"Sleeping for {sleep_duration} seconds")
```

blakejameson, 2025-04-30:
@Mikefly123 Thanks so much! and great idea

Mikefly123, 2025-05-17:
Closing this as it was completed with #243 !

closed 2025-05-17 by Mikefly123