proveskit/pysquared #241: [FEATURE] `make` install firmware

Mikefly123, 2025-04-22:
## Story
* **As a** pysquared developer
* **I want** a quick way to install the latest firmware build from the command line
* **So that** I don't have to download and then drag and drop the latest firmware.

## Acceptance Criteria

- [ ] A new `make` command that detects if there is an RP2XX0 chip attached via USB in Bootloader Mode and uploads a `firmware.uf2` file to it
- [ ] Takes arguments for specific firmware versions / release tags

## Technical Details
I feel like this is pretty straightforward. Right now we have stable versions of firmware in the [flight controller board repo](https://github.com/proveskit/flight_controller_board/tree/main/Firmware).

Eventually we probably want to create some kind of simple toolchain for developers to build firmware in real time and upload (because I think we eventually should consider directly compiling the `pysquared` library into firmware).

hrfarmer, 2025-05-03:
I want to tackle this, although I have a few questions:

1. Instead of having this in the pysquared repo, could this better fit being in the Makefile of each board repo? It's already where you have to be to install code to the board, so I think it'd make sense for the logic to install firmware for a specific board to also be there.
2. In the repo you linked, it seems to be missing firmware like the v4 board firmware, which I see is linked in the `README` in [pysquared](https://github.com/proveskit/pysquared/). Should I push that firmware to be in the `flight_controller_board` repo as well?

Mikefly123, 2025-05-03:
Hey @hrfarmer I actually concur that it would be a great idea to have this be a unique part of the `Makefile` for each individual board so we can directly target the latest firmware be default.

Thanks for the catch on the confusing naming for the v4 board firmware as well! I went ahead and just pushed a commit to make it clearer which version is the correct version.

closed 2025-05-09 by hrfarmer